OpenROV/openrov-electronics #31: Increase Copper Cuttout Hole on 2.8 Light Board

briangrau, 2015-09-27:
The M2 screws that are used to secure the camera board in place also go through the camera board and can cause the grounds to connect causing the lights to always be on and no user control. This seems to happen when the screws are tightened down a lot and is cutting though the coating on the light board plate.

briangrau, 2015-09-28:
Here is the forum thread where this problem was first brought to our attention.

https://forum.openrov.com/t/led-lights-on-all-the-time/3276

WaltHolm, 2015-10-10:
The copper pour in the area of the screws needs to be opened up so that the screw head cannot touch the copper.

briangrau, 2015-11-24:
Fixed in Light Board 2.8A

https://github.com/OpenROV/openrov-electronics/commit/51dcb5fd4651239b08779614ee4c81552e01517a

closed 2015-11-24 by briangrau